xperiments-in/xtouch #163: Is it possible to generate the config locally?

Dan-Ross, 2024-11-01:
## Feature Description

Not comfortable entering network credentials on someone elses' site. Can we make the config tool available for people to run locally, not just use the one you host?

## Justification

Security and trust

xperiments, 2024-11-02:
Go to my site, sabe Page as local. Disconnect internet and load the Page..

Btw i am stealing all your data in the esp32... We can be paranoic but my
content IS OpenSource... What better to secure things

Dan-Ross, 2024-11-03:
Thanks! Would make a great addition to the readme. I had looked for the config.html page in repo but didn't see it, so I didn't know if that page was open source too.

closed 2024-11-03 by Dan-Ross

dannutu, 2024-11-11:
Open source is all good but the man had a valid point though about "_Not comfortable entering network credentials on someone elses' site_".

Dan, you could also use the online page to download a config file with dummy values, just to have its structure. Then use any tool you like and you trust to locally encode your Wi-Fi SSID & password, email address and Bambu Lab "cloud" password and replace the placeholders in the file (just make sure you don't include CR / LF in those values).

But if you don't also check the x-touch source code yourself and build the image and allow it unrestricted internet access then you could as well just trust the author like he said :)