xperiments-in/xtouch #164: Fatal BBL JWT parsing error

FalcoVerus, 2024-11-04:
**Describe the bug**
As of this morning, the xtouch stopped working, after power on it's in an endless boot cycle, displaying messages:
---
Connecting
Connected
Processing JWT Token
Fatal BBL JWT parsing error
---
**To Reproduce**
Every boot, tried to reinstall and tried a new SD card as well.

**Expected behavior**
Work as before.

**XTOUCH (please complete the following information):**
 - FW: latest by the web installer

**Printers (please complete the following information):**
 - Device: P1S
 - FW Version: latest
-  AMS units: 1

Patrick075, 2024-11-04:
Same for me.
Also latest FW.
Receive at same time Verification Code Email from Bambu Lab !

sundyap, 2024-11-04:
Same problem here. Also received an verification code on email.

MissionTrooper-TK-76271, 2024-11-04:
Same here. I received an email verification cold via email from Bambu Labs.

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[XTouch][SCREEN] Setup
[   685][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x40114124 arg=0x3ffbfdb8
XTouch][SD] SD Card Type: [XTouch][SD] SDSC
[XTouch][SD] SD Card Size: 59MB
[XTouch][SD] Reading JSON file: /xtouch/settings.json
[XTouch][FS] Checking for touch config
[XTouch][TOUCH] Load from disk
[XTouch][SD] Reading JSON file: /config.json
[XTOUCH][CONNECTION] Connecting to WiFi ..
[XTOUCH][CONNECTION] Connected to the WiFi network with IP: 192.168.1.191
[XTouch][SD] Reading JSON file: /config.json
Getting accessToken from Bambu Cloud
deserializeJson() failed: EmptyInput
ets Jul 29 2019 12:21:46

rwejdling, 2024-11-04:
I'm getting this in the log:

```
Getting accessToken from Bambu Cloud
[ 26166][E][WiFiGeneric.cpp:1583] hostByName(): DNS Failed for api.bambulab.com
Received error: -1
deserializeJson() failed: EmptyInput
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
```

DNS resolves fine on my internal network:
```
➜ dig api.bambulab.com

; <<>> DiG 9.10.6 <<>> api.bambulab.com
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 30657
;; flags: qr rd ra ad; QUERY: 1, ANSWER: 2, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 4096
;; QUESTION SECTION:
;api.bambulab.com.              IN      A

;; ANSWER SECTION:
api.bambulab.com.       151     IN      A       172.64.152.215
api.bambulab.com.       151     IN      A       104.18.35.41
```

mpregione, 2024-11-04:
Same error here

davidmaurio, 2024-11-04:
Same issue.  Fatal BBL JWT parsing error. Bambu Labs sends Verification code via email each time it tries to connect. Tried rebooting xtouch and printer, logging into website, still no joy.

Gilles03300, 2024-11-04:
exactly the same problem for me. I installed the latest version in 0.9.140 still with the same problem

LucHeart, 2024-11-04:
Bambu probably added 2fa via email as a requirement now, not sure what is the best course of action here, maybe instead use local api and not cloud

umeiko, 2024-11-04:
For now, China region with phone number login not affected.

xperiments, 2024-11-04:
Fixed.

Reinstall from web

closed 2024-11-04 by xperiments

jsfrengeu, 2024-11-04:
> Fixed.
> 
> Reinstall from web

Is this the only fix? The screen is installed in my machine in a not so easy to remove way.

xperiments, 2024-11-05:
It's explained in the docs

https://github.com/xperiments-in/xtouch?tab=readme-ov-file#standard-firmware-update

LucHeart, 2024-11-05:
that wasnt the question tho? The question is, why did you mention web update specifically? Does OTA not work?

jsfrengeu, 2024-11-05:
Following the steps for the Standard Firmware Update did the trick.

xperiments, 2024-11-05:
will depend if it was previously activated

p0lfilt3r, 2024-11-05:
Works perfect with active OTA-Update. Thank you!