xperiments-in/xtouch #165: China can't login after updated 20241105's version

umeiko, 2024-11-05:
```
[XTOUCH][CONNECTION] Connecting to WiFi ..

[XTOUCH][CONNECTION] Connected to the WiFi network with IP: 192.168.1.106

[XTouch][SD] Reading JSON file: /config.json

Getting accessToken from Bambu Cloud

[ 46086][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-80) UNKNOWN ERROR CODE (0050)

Received error: -5
Requesting verification code from Bambu Cloud

Received error: 400
[XTouch][SD] Reading JSON file: /config.json

Getting accessToken from Bambu Cloud

[XTouch][SD] Writting JSON file: /xtouch/tokens.json

ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

```

umeiko, 2024-11-05:
But the older version login was okay

umeiko, 2024-11-05:
```
[XTOUCH][CONNECTION] Connected to the WiFi network with IP: 192.168.1.106

[XTouch][SD] Reading JSON file: /config.json

Getting accessToken from Bambu Cloud

[ 17441][V][HTTPClient.cpp:252] beginInternal(): url: https://api.bambulab.cn/v1/user-service/user/login

[ 17453][D][HTTPClient.cpp:263] beginInternal(): unexpected protocol: https, expected http

[ 17461][V][HTTPClient.cpp:252] beginInternal(): url: https://api.bambulab.cn/v1/user-service/user/login

[ 17471][D][HTTPClient.cpp:303] beginInternal(): protocol: https, host: api.bambulab.cn port: 443 url: /v1/user-service/user/login

[ 17483][D][HTTPClient.cpp:598] sendRequest(): request type: 'POST' redirCount: 0

[ 17518][V][ssl_client.cpp:62] start_ssl_client(): Free internal heap before TLS 180668

[ 17526][V][ssl_client.cpp:68] start_ssl_client(): Starting socket

[ 17587][V][ssl_client.cpp:146] start_ssl_client(): Seeding the random number generator

[ 17596][V][ssl_client.cpp:155] start_ssl_client(): Setting up the SSL/TLS structure...

[ 17606][D][ssl_client.cpp:176] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!

[ 17615][V][ssl_client.cpp:254] start_ssl_client(): Setting hostname for TLS session...

[ 17623][V][ssl_client.cpp:269] start_ssl_client(): Performing the SSL/TLS handshake...

[ 18289][V][ssl_client.cpp:290] start_ssl_client(): Verifying peer X.509 certificate...

[ 18297][V][ssl_client.cpp:298] start_ssl_client(): Certificate verified.

[ 18304][V][ssl_client.cpp:313] start_ssl_client(): Free internal heap after TLS 138464

[ 18312][D][HTTPClient.cpp:1170] connect():  connected to api.bambulab.cn:443

[ 18321][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 224 bytes...

[ 18333][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 47 bytes...

[ 18530][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'HTTP/1.1 200 OK'

[ 18537][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Date: Tue, 05 Nov 2024 09:47:44 GMT'

[ 18547][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Content-Type: application/json; charset=utf-8'

[ 18557][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Transfer-Encoding: chunked'

[ 18565][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Connection: keep-alive'

[ 18573][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'vary: Accept-Encoding'

[ 18581][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'vary: Origin'

[ 18588][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'b3: 31fe7df19bfe365c2fc08699b3a3e611-c08c615a0e8f6fdb-1'

[ 18599][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'traceparent: 00-31fe7df19bfe365c2fc08699b3a3e611-c08c615a0e8f6fdb-01'

[ 18612][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'x-request-id: 4f03fa0a-f681-441b-b2fd-5bfd31963ba0'

[ 18622][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'CF-Cache-Status: DYNAMIC'

[ 18631][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Set-Cookie: __cf_bm=J5_SKL8tELGWyfJmleNNp_UIHv8_bwX6KS1dIr.QIR0-1730800064-1.0.1.1-hTSF5Z5A0CnLZt3ksbMfRJ1hC1oFNtI.6xnLFevX8sJ7sTdLx.rirAwk5.2XwVi5OHrsaD_G2clfaEmYmoXypg; path=/; expires=Tue, 05-Nov-24 10:17:44 GMT; domain=.bambulab.cn; HttpOnly; Secure; SameSite=None'

[ 18661][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Strict-Transport-Security: max-age=15552000; includeSubDomains; preload'

[ 18674][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'expect-ct: max-age=86400, enforce'

[ 18683][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'referrer-policy: same-origin'

[ 18691][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'x-content-type-options: nosniff'

[ 18700][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'x-frame-options: SAMEORIGIN'

[ 18709][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'x-xss-protection: 1; mode=block'

[ 18717][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'Server: cloudflare'

[ 18725][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'CF-RAY: 8ddbec93e9069e33-FOC'

[ 18734][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: 'alt-svc: h3=":443"; ma=86400'

[ 18742][V][HTTPClient.cpp:1264] handleHeaderResponse(): RX: ''

[ 18748][D][HTTPClient.cpp:1321] handleHeaderResponse(): code: 200

[ 18754][D][HTTPClient.cpp:1328] handleHeaderResponse(): Transfer-Encoding: chunked

[ 18762][D][HTTPClient.cpp:642] sendRequest(): sendRequest code=200

[ 18768][D][HTTPClient.cpp:936] writeToStream():  read chunk len: 2114

[ 18788][D][HTTPClient.cpp:1460] writeToStreamDataBlock(): connection closed or file end (written: 2114).

[ 18799][D][HTTPClient.cpp:936] writeToStream():  read chunk len: 0

[ 18805][D][HTTPClient.cpp:388] disconnect(): still data in buffer (2), clean up.

[ 18813][D][HTTPClient.cpp:393] disconnect(): tcp keep open for reuse

[ 18819][D][HTTPClient.cpp:388] disconnect(): still data in buffer (2), clean up.

[ 18827][D][HTTPClient.cpp:393] disconnect(): tcp keep open for reuse

[ 18838][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.

[ 18846][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.

[XTouch][SD] Writting JSON file: /xtouch/tokens.json

[ 19159][V][WiFiGeneric.cpp:363] _arduino_event_cb(): STA Disconnected: SSID: 218-3, BSSID: 48:7d:2e:bb:72:99, Reason: 8

[ 19171][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED

[ 19173][V][WiFiGeneric.cpp:344] _arduino_event_cb(): STA Stopped

[ 19185][W][WiFiGeneric.cpp:1062] _eventCallback(): Reason: 8 - ASSOC_LEAVE

[ 19192][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 3 - STA_STOP

ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0030,len:1184

load:0x40078000,len:13232

load:0x40080400,len:3028

```

umeiko, 2024-11-05:
Seems like china's phone number login doesn't need for 2fa code then that causes this

xperiments, 2024-11-05:
We will wait some days to check if BBL changes also the way for China,
Question, are y from China, or just have a China Printer?
Can you come to discord? https://discord.gg/RytEDEgfR3

umeiko, 2024-11-05:
I am in China, and have bambu printers login in China's bambu account.

xperiments, 2024-11-05:
Can you come to Discord?
Do you have access from China?

xperiments, 2024-11-05:
ATM

Download 

https://github.com/xperiments/xtouch-bin/blob/b687716833484012c14113038bfe6a88a4e24481/fw/firmware.bin

Put at SD root, reboot screen

This will install the prev version that was working for y

Remember to disable OTA

umeiko, 2024-11-06:
> Can you come to Discord? Do you have access from China?
> […](#)

Yes， I have access from China，I have entered Discord channel

Sebba1976, 2024-11-06:
I have a similar issue after this update and reverting back to old firmware produces a fatal BBL JWT Parsing error

umeiko, 2024-11-06:
> I have a similar issue after this update and reverting back to old firmware produces a fatal BBL JWT Parsing error

you can try re-format your tf card and rewrite a new config.json

umeiko, 2024-11-07:
I tried accuired a code from bambu web login page, and put it into the webpage from esp32. then i got this and reboot forever:

[image]
```
[XTOUCH][CONNECTION] Connected to the WiFi network with IP: 192.168.1.110

[XTouch][SD] Reading JSON file: /tokens.json

184888

[XTouch][SD] Reading JSON file: /tokens.json

Getting device list from Bambu Cloud

Received error: 401
[XTouch][SD] Reading JSON file: /printer.json

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```

umeiko, 2024-11-07:
I even found that. if I load the _auth_token from serial and use it to postman, it returns 200

[image]

[image]

but recives 401 from esp32 very strange.

umeiko, 2024-11-07:
aother type of error:
```
[XTouch][SD] SD Card Size: 7580MB
[XTouch][SD] Reading JSON file: /xtouch/settings.json

[XTouch][FS] Checking for touch config

[XTouch][TOUCH] Load from disk

[XTouch][SD] Reading JSON file: /config.json

[XTOUCH][CONNECTION] Connecting to WiFi ..

[XTOUCH][CONNECTION] Connected to the WiFi network with IP: 192.168.1.110

[XTouch][SD] Reading JSON file: /xtouch/tokens.json

157120

[XTouch][SD] Reading JSON file: /xtouch/tokens.json

Getting device list from Bambu Cloud

[ 11192][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -1

Received error: -1
[XTouch][SD] Reading JSON file: /xtouch/printer.json

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:

PC      : 0x400d3a2f  PS      : 0x00060f30  A0      : 0x800d3a51  A1      : 0x3ffb2060  

A2      : 0x00000001  A3      : 0x3f4033a2  A4      : 0x00000000  A5      : 0xffffff80  

A6      : 0x00000000  A7      : 0x3ffb1eec  A8      : 0x3f41ec58  A9      : 0x3ffb2000  

A10     : 0x00000000  A11     : 0x3ffb2068  A12     : 0x00000011  A13     : 0x3ffb2110  

A14     : 0x00000000  A15     : 0x3ffe4d5c  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  

EXCVADDR: 0x00000009  LBEG    : 0x4008a568  LEND    : 0x4008a57e  LCOUNT  : 0x00000000  

Backtrace: 0x400d3a2c:0x3ffb2060 0x400d3a4e:0x3ffb2080 0x400dc652:0x3ffb20a0 0x400dda05:0x3ffb2240 0x4011a03a:0x3ffb2290
```

umeiko, 2024-11-08:
i made up a pr to fix this : `https://github.com/xperiments-in/xtouch/pull/170`

manith008, 2024-11-08:
how to install it

umeiko, 2024-11-08:
> how to install it

compile sourcecode from my fork.

dannutu, 2024-11-10:
> Download
> 
> https://github.com/xperiments/xtouch-bin/blob/b687716833484012c14113038bfe6a88a4e24481/fw/firmware.bin
> 
> Put at SD root, reboot screen
> 
> This will install the prev version that was working for y
> 
> Remember to disable OTA

I also wanted to try this version since the current one doesn't work without access to Bambu cloud.

the ESP went through the update, I could see the % increasing and then it briefly said "update error" and it was back to the "Invalid BBL Cloud credentials" dead end :(

The serial log shows "Error Reading JSON File: /xtouch/ota.json". I suppose this is because it was never created, since I was never able to use xtouch.

Do you have a "full" 0.9.140 version that I could download and flash with esptool, just like the current xtouch.web.0.9.149.bin from the ESP web tools manifest? (i.e. not just the OTA part of it) Thanks.

xperiments, 2024-11-11:
@dannutu Thats not related to china ccounts

closed 2024-11-11 by xperiments

dannutu, 2024-11-11:
Yes, thanks, I eventually figured that out :)
I thought I'd also have a look at the **closed** issues (not shown by default) and so I found issue #130 
The only thing now is... what is the structure of the old wifi.json file? :) I can't seem to find it.