xperiments-in/xtouch #166: Message to go to http://(local ip) - which then asks for a Bambu Lab Verification Code

Sebba1976, 2024-11-05:
As of yesterday my XTOUCH is asking me to go to http://(local ip) - which then asks for a Bambu Lab Verification Code - Check your email for code.

The problem is that I haven'
[image]
t received an email with a code.

Captain0Kangaroo, 2024-11-06:
I'm also getting this issue as of 2 days ago.

tuc0w, 2024-11-07:
I just updated to the current version, got the msg and also received the mail.
Did you check your Bambu account that it has the correct email address set? Maybe check your junk folder?

umeiko, 2024-11-07:
china has no email and have to login with phone number, which can't recive the message

dannutu, 2024-11-10:
Hmmm, it seems that I missed the x-touch version that also worked in local LAN mode by just a week... the current one clearly wants to connect to Bambu's "cloud" servers. Which I'm having none of, sorry.

xperiments, 2024-11-11:
China bug resolved

closed 2024-11-11 by xperiments