xperiments-in/xtouch #168: Setup without BambuLab Account

Mqxx, 2024-11-07:
How can I setup the Xtouch without a BambuLab account? I use my printer in Lan Only Mode and I don't want all my data beeing send though or to BambuLab. Is it even possible to setup without account?

dannutu, 2024-11-10:
I heard about x-touch some time ago from "Teaching Tech" Michael and I finally got some time today to see if I can make it work with my printer. I do not use the Bambu "cloud" though - everything has to work and stay local.

In the "xtouch Bambu Lab P1P/P1S touch screen upgrade guide" video (which was published about a year ago) at around 6m38 Michael clearly said:

> 6:37 [...] the
> 6:39 unit will now **search for printers on**
> 6:41 **your local network and list them for you**
> 6:42 **to connect to** 

Even the readme file seems to confirm this:

> ### Linking Printer
> 
> During the installation process, you will need to link your xtouch screen with your 3D printer. Follow these steps to successfully establish the connection:
> 
> 1. **Searching for Printers**:
> 
>    - **After powering on the touch screen, you will be greeted with a "Searching for printers" screen. The touch screen will scan for available printers**.

So I was very hopeful this would work fine in LAN mode.

I therefore installed the x-touch firmware image on one of those CYD ("cheap yellow display") ESP32 devices, I (manually) created the config.json file, wrote it on a micro SD card, inserted the card in the slot and powered up the device. Two pen touches on the screen to calibrate it and a reboot later the device successfully connected to Wi-Fi (on the very same access point & subnet as the P1 printer). But since that network doesn't have access to the public internet the device got stuck displaying an "Invalid BBL Cloud credentials" message.

I can see that right after it booted up it connected to Wi-Fi, made a DNS query to resolve the hostname api.bambulab.com and then it tried 3 times in quick succession to connect to the https port on one of the public IP addresses returned by the DNS server. All 3 attempts were unsuccessful since that network doesn't have access to the public internet. The serial log showed "Requesting verification code from Bambu Cloud" and then 3 "Connection reset by peer" socket-related errors.

It then started its own http server and advertised it via mDNS as xtouch.local. Connecting to it via http brings up a very simple page saying "Enter Bambu Lab Verification Code / Check your email for the code". 

And after a quick look though the source code I think that the current version of x-touch only works with Bambu cloud. I don't know anything about Bambu's "cloud" api and I'm not interested to find out but it seems to me there is no "scan" on the local network, just a download once connected to Bambu's servers.

For example - this is where my device is stuck (code from cloud.hpp):

`
  bool login(String verificationCode = "")
  {

    lv_label_set_text(introScreenCaption, LV_SYMBOL_CHARGE " Login BBL Cloud");
    lv_timer_handler();
    lv_task_handler();
    delay(32);

    DynamicJsonDocument wifiConfig = xtouch_filesystem_readJson(SD, xtouch_paths_config);

    if (!wifiConfig.containsKey("cloud-region") || !wifiConfig.containsKey("cloud-email") || !wifiConfig.containsKey("cloud-password"))
    {

      lv_label_set_text(introScreenCaption, LV_SYMBOL_CHARGE " Invalid Bambu Cloud credentials");
      lv_timer_handler();
      lv_task_handler();
      delay(3000);
      ESP.restart();

      return false;
    }

    _region = wifiConfig["cloud-region"].as<const char *>();
    _email = _decodeString(wifiConfig["cloud-email"].as<String>());
    _password = _decodeString(wifiConfig["cloud-password"].as<String>());

    lv_label_set_text(introScreenCaption, LV_SYMBOL_CHARGE " Processing JWT Token");
    lv_timer_handler();
    lv_task_handler();
    delay(32);

    if (verificationCode == "")
    {
      _auth_token = _getAuthToken();
    }
    else
    {
      _auth_token = _getAuthToken(verificationCode);
    }

    if (_auth_token == "verifyCode")
    {
      String gotoCode = "Go to http://" + WiFi.localIP().toString();
      lv_label_set_text(introScreenCaption, gotoCode.c_str());
      lv_timer_handler();
      lv_task_handler();

      return false;
    }

    if (_auth_token == "")
    {
      lv_label_set_text(introScreenCaption, LV_SYMBOL_CHARGE " Invalid BBL Cloud credentials");
      lv_timer_handler();
      lv_task_handler();
      delay(3000);

      return false;
    }
`

xperiments, 2024-11-11:
https://github.com/xperiments-in/xtouch/issues/130

closed 2024-11-11 by xperiments

Mqxx, 2024-11-14:
> #130

I did this and it worked fine, but this is not the latest version. I am not sure how much the cloud is "baked into" the current OS but I hope to see an option in the future where I can skip all the cloud stuff.

xperiments, 2024-11-14:
@Mqxx ATM I don't plan to update things in 2.8

BBL is changing things right now... ATM We stay in cloud.....

"[...] as I use cloud for control printer, not for sending print files... they remain local...". 

It remains true, you just need to wait for the 5 inch to get file accesss and this way is a CLOUD-NOCLOUD

CLOUD for controlling printer
NOCLOUD for storing files in the screen SD

Also 5 inch comes w local support

Later when I have time I will port 5 inch to 2.8 (the features that fits)

dannutu, 2024-11-14:
Thanks for sharing the plan. I'm not sure though that I understood what exactly is going to work entirely local and what is going to rely on the "cloud". But I'm pretty happy with the old 0.9.58 version, the only additional feature that I'd _really_ like to have  would be a "start print from uSD card model" capability (i.e. from the printer's built-in uSD card reader). I have no issues whatsoever to transfer the models from the computer to the uSD card in the printer using ftps (and in a certain way I quite like it) but I'd really like to also have an option to start them without having to flip up the xtouch and use the built-in 1970's LCD panel

xperiments, 2024-11-14:
2.8 is not capable of managing the Filebrowser system... forget about it... only 5inch

dannutu, 2024-11-14:
Yes, I imagined that displaying multiple files on the smaller screen might be (very) difficult.

But do you know how to start a print job via MQTT if you already know the name of the model file on the uSD card? Is it even possible to start a print job from the uSD card using MQTT? If you know how to start it I could simply use an MQTT client on the computer to publish the required message.

Or how about including just a "Print" button on the small CYD screen that would always try to print a file with a fixed name (e.g. 00PRINTME.gcode) from the uSD card? And you'd simply have to overwrite that file if you wanted to print something else.

Mqxx, 2024-11-15:
Currently I have both, the 2.8 inch and 5 inch display. I use OrcaSlicer to start a  print job aka sending files to the printer. I think many people do so. I don't need the capability to print from the SD Card of the display. I just don't want to be forced to use the cloud for simple setup. It would be just fine if I could just skip the step for setting up the cloud and move on.

dannutu, 2024-11-15:
Hey there, I didn't mean to say that _you_ needed the capability to start a print from xtouch, I am the one who would need that :) I only mentioned it because you wrote "_but this is not the latest version_" - as I said above, I am pretty happy even with the old 0.9.58 xtouch version, not sure what you are missing compared with the current version? 

But yes, I agree that it would be great if the current version could "_just skip the step for setting up the cloud and move on_", however I suspect it's not as simple as that - it could for example be that in the current version (plenty of) info is retrieved from the cloud api that cannot be retrieved locally in the LAN mode. I'm speculating here, as I don't have the time to go through the code to be sure, but I guess this is why the author wrote "_ATM We stay in cloud_", "_I use cloud for control printer_" and "_CLOUD for controlling printer_".

(BTW, I would like to have the capability to start a print job from xtouch because I am running Orca on a Raspberry Pi and there is no Bambu Studio or Orca network plugin for Linux running on ARM)

Mr-Savag3, 2025-01-17:
This is something I've been hoping for for ages. I don't need to use the screen to start prints, I just would like it for controlling the printer more easily. I very rarely start a print from the SD card, but I would love for my 2.8in screen to be able to control everything else.

Mqxx, 2025-01-20:
Stay in cloud didn't age well...