xperiments-in/xtouch #171: not working since the last update

poli27, 2024-11-12:
Connect to http://x.x.x.x 
the display connection stops here... the two-step identification is not activated in the bambu account, and yet confirmation codes are coming, what is going on and what is the solution to this?

xperiments, 2024-11-12:
Connect to http://x.x.x.x/

closed 2024-11-12 by xperiments

poli27, 2024-11-12:
the x indicates the IP address, I just typed it in... that's how far the display goes!

Fredzp1s, 2024-11-20:
Hello, I have install Xtouch on board this day and it don't connect to bambu printer, only say Go to HTTP:/192.168.1.10
What does i made after ? is there a bambu restriction ?

xperiments, 2024-11-20:
@Fredzp1s You need to goto the address it shows in the screen with a browser

Check your email... you will receive a bbl verification. code

Them enter this code in the page

PD: If you received a lot of codes today by ex... Wait some hours as they maybe blocked temporally you the access