xperiments-in/xtouch #172: Bambu Changed Auth_token format in China, China can't login since 2024.11.13, returns `Fatal BBL JWT parsing error`

umeiko, 2024-11-15:
As Titled.
All we know is `/src/xtouch/cloud.hpp : 92` 
```C
String _getUserFromAuthToken()
```
Is not working in China, which returns a `" Fatal BBL JWT parsing error"`

I made a temp fix in my fork, but should be a better way to fix this

hujingwen1025, 2024-11-16:
Hi, is there a compiled .bin version of this?

umeiko, 2024-11-16:
The best method is to get printing history information from this website 

`https://bambulab.cn/api/v1/iot-service/api/user/project` 

and extract the user-id from the returned JSON, but my technical ability is insufficient. There are several issues in the process 
1. the data obtained is a complex JSON structure, and I am not familiar with how to extract the required content from it. 
2.  the data is extremely large, and I am unsure if the ESP32's RAM is sufficient to process this data. 
3.  if a new user has never printed anything, the user-id will not exist in the data, and we need to handle such special cases.

umeiko, 2024-11-16:
> Hi, is there a compiled .bin version of this?

This is just a temporary solution for my own use; it's not perfect, so I don't plan to publicly release it.

looxonline, 2024-11-16:
> The best method is to get printing history information from this website
>
> `https://bambulab.cn/api/v1/iot-service/api/user/project`
>
> and extract the user-id from the returned JSON, but my technical [...]

https://api.bambulab.cn/v1/user-service/my/profile

umeiko, 2024-11-16:
> https://api.bambulab.cn/v1/user-service/my/profile

this works! may be the best solution

santanig5, 2024-11-18:
Is there a workaround for non technical U.S. users?

travellingman9999, 2024-11-20:
This same "Fatal BBL JWT parsing error" issue has now occurred for me, using an EU/world registered account. Running latest firmware on Bambu P1S and current (as of 1530 Eastern European Time) version of XTouch .bin. 

I've repeatedly tried the XTouch web installer and manual update processes to no effect.

I'm not a coder, but I'm technically proficient and am happy to investigate further if anyone can point me in the right direction.

Sneard, 2024-11-20:
Can confirm "Fatal BBL JWT parsing error" as user in EU/world.

`rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[XTouch][SCREEN] Setup
SPI3: sck:14, miso:-1, mosi:13, ss:-1
1111SPI3: sck:14, miso:-1, mosi:13, ss:-1
SPI2: sck:25, miso:39, mosi:32, ss:33
1111SPI2: sck:25, miso:39, mosi:32, ss:33
[   670][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x40118cec arg=0x3ffc0254
SPI3: sck:18, miso:19, mosi:23, ss:5
1111SPI3: sck:18, miso:19, mosi:23, ss:5
XTouch][SD] SD Card Type: [XTouch][SD] SDHC
[XTouch][SD] SD Card Size: 122240MB
[XTouch][SD] Reading JSON file: /xtouch/settings.json
[XTouch][FS] Checking for touch config
[XTouch][TOUCH] Load from disk
[XTouch][SD] Reading JSON file: /config.json
[XTOUCH][CONNECTION] Connecting to WiFi ..
[XTOUCH][CONNECTION] Connected to the WiFi network with IP: 192.168.178.75
[XTouch][SD] Reading JSON file: /config.json
Getting accessToken from Bambu Cloud
Requesting verification code from Bambu Cloud
[XTouch][SD] Reading JSON file: /config.json
Getting accessToken from Bambu Cloud
Starting webserver...
Webserver started
http://xtouch.local
192.168.178.75
Received verification code: 123456
[XTouch][SD] Reading JSON file: /config.json
Getting accessToken from Bambu Cloud
Failed to decode base64
ets Jul 29 2019 12:21:46`

santanig5, 2024-11-21:
I opened a case with Bambu, they know they broke it, but since it's unsupported they don't care, nor are they planning to offer a touch screen option.

xperiments, 2024-11-21:
Can y share his answer please

santanig5, 2024-11-21:
[image]

stet345434543454345, 2024-11-22:
I'm also having the same issue. I'm in the us region and it continued to work until i had to restart my printer/xtouch, then it stopped working.

RAHoebe, 2024-11-23:
Same problem here from europe netherlands

xperiments, 2024-11-25:
Closed at [07adcb2](https://github.com/xperiments-in/xtouch/commit/07adcb2aff618993ed91f6070dae20afc8af1dbb)

closed 2024-11-25 by xperiments

santanig5, 2024-11-25:
When I entered the code successfully I now see a fatal parsing error.

santanig5, 2024-11-25:
Yes, have it working now.

Thank you very much!

Sent from my Verizon, Samsung Galaxy smartphone

@santanig5<https://github.com/santanig5> have you read?

Faasteer, 2024-11-26:
@xperiments Oh man, thank you so much for all your hard work on this project! 🙏 Even if directly BambuLab does all these problems. My P1S wouldn't be complete without the Xtouch display, the few days it didn't work was hell on earth. 🤣

[image]