xperiments-in/xtouch #175: Invalid BBL Cloud credentials

WinstonLeWolf, 2024-11-23:
**Describe the bug**
The XTouch fails to start due to invalid BBL Cloud credentials.  I receive an email verification code.  Going to the Xtouch IP address and manualing entering the token does not fix it. 

**To Reproduce**
Power on the device

**Expected behavior**
The Xtouch to provide me the option to connect to my printer. 

**Screenshots**
[image]

**XTOUCH (please complete the following information):**
 - FW: 0.9.149

Tried installing the firmaware.bin as well as the web installer

**Printers (please complete the following information):**
 - Device: P1P
 - FW Version 01.06.01.02
-  AMS units: 00.00.06.44

**Additional context**
Add any other context about the problem here.

ahernandez1801, 2024-11-23:
Getting the same error, but also receiving "base64 decode failed" after inputting the verification code

ScottTheHippy, 2024-11-24:
I'm having this as well after updating the P1S firmware.  I've nuked and reinstalled the Xtouch and when it boots, I can see the error 429 in the logs as well, and the "invalid BBL Cloud credentials" on the display.  I can load the IP of the Xtouch in a browser, and feed it the six digit auth code from Bambu and it'll say "Verification code submitted successfully!" but nothing changes.

rubyredrick, 2024-11-25:
I'm seeing the exact same things.  I accepts the code, but i then get the invalid BBL Credentials message.  I also see the failed to decode base64 in the logs.

xperiments, 2024-11-25:
Closed at [07adcb2](https://github.com/xperiments-in/xtouch/commit/07adcb2aff618993ed91f6070dae20afc8af1dbb)

WinstonLeWolf, 2024-11-25:
Thank you.  I started completely from scratch using the new installation instructions.  I was getting an MQTT error upon boot.  After getting static IPs for both the Xtouch and my printer, the issue appears to have been solved.

ScottTheHippy, 2024-11-26:
This extension isn't working for me.  I'm logged in to BambulLab.com (in Chrome), have tried disabling/re-enabling the extension, logged out and back in, and restarted the browser, and I keep getting "Log into BambuLabs, then re-open the extension to continue".
[image]

ScottTheHippy, 2024-11-26:
I had to go grab a Windows system and dust it off to get your extension to work.  The prior feedback and screenshot was on a Mac.  On a Windows host, the extension worked as described.

rubyredrick, 2024-11-26:
Windows is not an option for me.

IIRC the Xtouch display say theres no config.json file, but i guess the chrome extension is supposed to replace/create that.  As i said it looks like it's not even trying to obtain an ip address from the router.

xperiments, 2024-11-26:
@ScottTheHippy It was developed on a MAC so I can confirm it works

xperiments, 2024-11-26:
If you can come to Discord https://discord.gg/yRNWJj55 can be great... I will be not too much more today... like 1h from now....

xperiments, 2024-11-26:
Also, can you download the lattest extension version... https://xperiments.in/xtouch-bin/extensions/xtouch28.zip