xperiments-in/xtouch #176: Still Can't provision

rubyredrick, 2024-11-25:
I followed the new installation process using the Chrome extension in the recently updated README.  The extension fails with the message:

          An error occurred provisioning the device. Please check that the
          entered IP is correct
        
For what it might be worth, my SSID is a space separated list of words.  The device is not showing up in my routers list of attached devices, so it looks like it never made a DHCP request.

xperiments, 2024-11-25:
Read the instructions carefully. Is well explained there....

closed 2024-11-25 by xperiments

rubyredrick, 2024-11-26:
Sorry, but there's another issue opened by another user, the extension doesn't seem to work on Mac

xperiments, 2024-11-26:
@rubyredrick come to discord, show me... Does not "work" is not an issue...

- What
- When
- Where

You (all) think I am magician and know the bugs you have beforehand...
It is developed in MAC, tested on other macs.... So I need more info to be able to help