xperiments-in/xtouch #178: create provisioning. json

tbc0309, 2024-11-27:
I am still unable to use Chrome Extension version 1.3.2, so I manually created provisioning. json and tested OK

{
  "ssid": "XXX",
  "pwd": "12345678",
  "cloud-email": "xxx",
  "cloud-region": "China", #
  "cloud-username": "u_12345678",
  "cloud-authToken": "xxx"
}

cloud-region  is China or World
cloud-email is phone number(China )、email(World )

cloud-username is u_uidStr

[image]

cloud-authToken

[image]

Quacker01-sys, 2024-11-27:
Hi!

I could really use some hand-holding on this stuff. I'm using our P1S as a production machine in a small business. We love the Xtouch screen but we foolishly updated our P1S and now it's not working. I've followed what you did with the provisioning file up until I got to "cloud-authToken" I don't even know what browser page that is so I can't find the token to finish the file. Can you please help? TIA!

xperiments, 2024-11-27:
@Quacker01-sys come on discord and give more info

Midyew59, 2024-11-28:
Attempted to follow your guide.  uidstr is not returning any results when searched for.

tbc0309, 2024-11-29:
> Attempted to follow your guide. uidstr is not returning any results when searched for.

login then open:

World https://bambulab.com/api/v1/iot-service/api/user/project

China  https://bambulab.cn/api/v1/iot-service/api/user/project

Midyew59, 2024-11-29:
That got me a little further!

Now the screen says to provision at the IP address I would expect.  Upon visiting that IP address though nothing happens.  Just the Xtouch logo appears on the screen.  But, your information helped make some progress.  I just gotta sit down and start at square one now I think.  Man this would be much easier if the darn Extension would work like it is supposed to lol.

Thanks for the help!

tbc0309, 2024-11-29:
> That got me a little further!
>
> Now the screen says to provision at the IP address I would expect. Upon visiting that IP address though nothing happens. Just the Xtouch logo appears on the screen. [...]

This is the operation according to the developer's Google Chrome Extension process, but the extension is currently unavailable, so for now, we will manually create provisioning. json ourselves.

xperiments, 2024-11-29:
@tbc0309 The extension is avaliable is working for other china ppl

xperiments, 2024-11-29:
Fixed. Update your extensions

closed 2024-11-29 by xperiments

Haggy38, 2024-12-04:
No idea what to do, Im on Mac

Haggy38, 2024-12-04:
cloud-email is phone number(China )???