xperiments-in/xtouch #181: Xtouch Extension loading indefinitely.

Midyew59, 2024-11-28:
**Describe the bug**
Xtouch Extension not functioning.

**To Reproduce**
Click on extension in chrome.  Loads indefinitely.

**Expected behavior**
Connect to Bambulab.com and begin asking for pertinent information.

**XTOUCH (please complete the following information):**
Most recent version.  Updated OTA and the device began asking for provisioning.json

**Printers (please complete the following information):**
 - Device: P1S
 - FW Version Most Current

**Additional context**
I've been through the readme, read all the pertinent solved and unsolved issues to find answers and found nothing that helps.  Yes I am logged into Bambulab.com, on the webpage when launching the extension,  Using the current version of Chrome and the newest version of the extension.  Attempted to join discord server but the invite did not work.

xperiments, 2024-11-29:
https://discord.gg/yRNWJj55

Midyew59, 2024-11-29:
New extension update allowed everything to work perfectly.  Everything is all good!

closed 2024-11-29 by Midyew59