xperiments-in/xtouch #184: Online installer

Marc579, 2024-12-11:
Hallo,

Ich habe ein 5" Display gekauft. Leider funktioniert die Online Installation nicht. Esp32 s3 Not supported. Was mache ich verkehrt?

xperiments, 2024-12-11:
5inch is in beta, a open beta will be out in 1-2w

closed 2024-12-11 by xperiments

Marc579, 2024-12-11:
Tanks dir replyGesendet mit der WEB.DE iPad AppAm 11.12.24 um 12:23 schrieb Pedro Casaubon
                
            
            
                
                    Von: "Pedro Casaubon" ***@***.***>Datum: 11. Dezember 2024An: "xperiments-in/xtouch" ***@***.***>Cc: "Marc579" ***@***.***>,"Author" ***@***.***>Betreff: Re: [xperiments-in/xtouch] Online installer (Issue #184)
                
                
5inch is in beta, a open beta will be out in 1-2w

xperiments, 2024-12-11:
?