xperiments-in/xtouch #191: Unable to Boot after install - UNABLE TO DETERMINE TIME, RESTARTING

Neil-Birtles, 2025-02-09:
**Describe the bug**
Once installed and Provisioned, I am unable to get into the device. Just sits on the first page with the SD card and Wifi Symbol
Logs tell me its the time server and that the connection to the wifi goes through

**To Reproduce**
Install xtouch pro using https://xperiments.in/xtouch-pro/
Generate config.json and provisioned via Chrome Extension
Power device (JC8048W550)

**Expected behavior**
Expect device to power on and load xtouch interface

**XTOUCH (please complete the following information):**
 - FW: [e.g. 1.0.0]

**Printers (please complete the following information):**
 - Device: P1S

**Additional context**
Attempted to re-install multiple times along with manually adding the config.json to the SD Card. nothing seems to be working. 

Logs

[XTouch][WiFi] Connecting to xxx
.
[XTouch][WiFi] Connected to xxx

[XTouch][WebServer] Starting webserver...
[XTouch][WebServer] Starting mDNS with name: xtouch-kopesases.local at IP: 192.168.xx.xx
...
[XTouch][NTPTime] Starting NTP time sync
[-][-][-][-][-][-][-][-][-][-]
[XTouch][NTPTime] Timezone: Europe/London
-----[XTouch][NTPTime] Setting timezone to: Europe/London
[XTouch][NTPTime] Timezone set to: GMT0BST,M3.5.0/1,M10.5.0
[XTouch][NTPTime] Failed to obtain time
UNABLE TO DETERMINE TIME, RESTARTING

xperiments, 2025-02-09:
That is mostly because your net can't access the NTP server at
ntp.google.com

Try to use your phone as a hotspot and check if there it connects... if so
is something wifi related...

Cheers!!

Neil-Birtles, 2025-02-10:
You're correct that this works with hotspot, Although through all my efforts I cannot get this to connect via my Wifi Router,

I've gone through everything on the Router itself, Forwarded port 123 and yet nothing. Even been on the phone to my ISP (Vodafone).

xperiments, 2025-02-11:
Come to discord please

https://discord.gg/Fv9pxchK

sonicmingit, 2025-03-24:
> It seems like I'm having the same issue as you—are you also using the device in China?

同样的问题,在中国使用.
主要需要解决设备的 time.google.com 访问问题.
我开始也是,表现为一直连不上wifi,提示错误,看这个issue,连接输出日志发现同样问题.
搞定 time.google.com 的访问就可以了.