xperiments-in/xtouch #192: Xtouch-Pro unable to connect to WiFi

teknoziz, 2025-02-19:
**Describe the bug**
First time install, unable to connect to primary or IoT network with correct credentials. Primary broadcasting 2.4, 5, 6. Even tried disabling 5 and 6. Multiple network refreshes and reboots.

**To Reproduce**
Steps to reproduce the behavior:
1. Scans for networks
2. Click on either my primary or IoT network
3. Enter password
4. See error "Failed to connect..."

**Expected behavior**
Connect to the network.

**Screenshots**
N/A

**XTOUCH (please complete the following information):**
 Whatever is current to 2/19/25

**Printers (please complete the following information):**
N/A

**Additional context**
N/A

teknoziz, 2025-02-20:
Today I tried the procedure on my PC instead of my iMac and I was able to connect to WiFi using the IoT 2.4 connection. The PC was also located near the printer and the cable modem, so not sure if that had anything to do with it. Was also able to connect locally and through the BBL cloud. SD card and Preferences don't seem to be accessible.

teknoziz, 2025-02-21:
I am guessing because I have the free version that the additional features are unavailable. Any idea when they will become available?

closed 2025-02-21 by teknoziz

xperiments, 2025-02-21:
No, we are in beta all is open ATM

teknoziz, 2025-02-21:
> xperiments
>  left a comment
> (xperiments-in/xtouch#192)
> No, we are in beta all is open ATM

Here is a screenshot of the screen, notice two options SD Card and Preferences greyed out.

￼

xperiments, 2025-02-21:
Yes, we are in beta, is not developed not a question of pro or not.

Come to discord, will be better....

https://discord.com/invite/Jdvp5CShnt

teknoziz, 2025-02-22:
Already subscribed, thanks! What is the best area to be notified of new releases? Will they be OTA upgrades or will it be another manual install?Sent from my iPad

https://discord.com/invite/Jdvp5CShnt