xperiments-in/xtouch #200: Connecting/Connected loop

sorz2122, 2025-04-15:
Hey, I have a problem: I’m using the 2.8 inch display.
After successfully installing the Google extension, flashing the screen, and adding the file to the SD card, the first boot shows “connecting,” then after a few seconds “connected,” then the SD card symbol flashes for a millisecond—and then it restarts. It seems to be stuck in some kind of boot loop.
Is there a solution for this?

Blazzycrafter, 2025-04-15:
here too...
but i connected the device to my pc and used tzhe installer to get the logs:
```
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[XTouch][SCREEN] Setup
SPI3: sck:14, miso:-1, mosi:13, ss:-1
1111SPI3: sck:14, miso:-1, mosi:13, ss:-1
SPI2: sck:25, miso:39, mosi:32, ss:33
1111SPI2: sck:25, miso:39, mosi:32, ss:33
[   685][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x40117904 arg=0x3ffc0250
SPI3: sck:18, miso:19, mosi:23, ss:5
1111SPI3: sck:18, miso:19, mosi:23, ss:5
XTouch][SD] SD Card Type: [XTouch][SD] SDHC
[XTouch][SD] SD Card Size: 29823MB
[XTouch][SD] Reading JSON file: /xtouch/settings.json
[XTouch][FS] Checking for touch config
[XTouch][TOUCH] Load from disk
[XTouch][SD] Reading JSON file: /provisioning.json
[XTOUCH][CONNECTION] Connecting to WiFi ..
[XTOUCH][CONNECTION] Connected to the WiFi network with IP: 192.168.178.XXX
Starting webserver...
Webserver started
http://xtouch.local
192.168.178.140
[XTouch][SD] Reading JSON file: /provisioning.json
132616
[XTouch][SD] Reading JSON file: /provisioning.json
[XTouch][SD] Reading JSON file: /provisioning.json
World
Getting device list from Bambu Cloud
Received error: 403
[XTouch][SD] Reading JSON file: /xtouch/printer.json
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d3667  PS      : 0x00060f30  A0      : 0x800d3689  A1      : 0x3ffb2050  
A2      : 0x00000884  A3      : 0x3f403ac9  A4      : 0x00000000  A5      : 0xffffff80  
A6      : 0x00000000  A7      : 0x3ffb1edc  A8      : 0x3f41f290  A9      : 0x3ffb1ff0  
A10     : 0x00000000  A11     : 0x3ffb2058  A12     : 0x00000011  A13     : 0x3ffb20f0  
A14     : 0x00000000  A15     : 0x3fff4a9c  SAR     : 0x00000011  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000088c  LBEG    : 0x4008a568  LEND    : 0x4008a57e  LCOUNT  : 0x00000000  

Backtrace: 0x400d3664:0x3ffb2050 0x400d3686:0x3ffb2070 0x400db7d9:0x3ffb2090 0x400dc8dd:0x3ffb2220 0x40118d06:0x3ffb2290

ELF file SHA256: 982b0cca12211e15```

nboggess18, 2025-04-16:
I am experiencing the same error. Please see abbreviated log below:

`Getting device list from Bambu Cloud
Received error: 403
[XTouch][SD] Reading JSON file: /xtouch/printer.json
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d3667  PS      : 0x00060f30  A0      : 0x800d3689  A1      : 0x3ffb2050  
A2      : 0x00000884  A3      : 0x3f403ac9  A4      : 0x00000000  A5      : 0xffffff80  
A6      : 0x00000000  A7      : 0x3ffb1edc  A8      : 0x3f41f290  A9      : 0x3ffb1ff0  
A10     : 0x00000000  A11     : 0x3ffb2058  A12     : 0x00000011  A13     : 0x3ffb20f0  
A14     : 0x00000000  A15     : 0x3fff5190  SAR     : 0x00000013  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000088c  LBEG    : 0x4008a568  LEND    : 0x4008a57e  LCOUNT  : 0x00000000  

Backtrace: 0x400d3664:0x3ffb2050 0x400d3686:0x3ffb2070 0x400db7d9:0x3ffb2090 0x400dc8dd:0x3ffb2220 0x40118d06:0x3ffb2290`

xperiments, 2025-04-17:
We have moved to LAN ONLY Mode as BBL seems clossing his Cloud for us.

Check the doc for updated instructions

closed 2025-04-17 by xperiments