xperiments-in/xtouch #201: Update appears to have caused an issue with the Touch Screen

Flarts001, 2025-04-17:
I turned on my Bambu Labs Printer today to start a print, and noted that the XTouch screen was indicating that it was downloading an update. The update downloaded successfully, showed that it was installing the update, but on restarting the screen, it's locked on the XTouch Splash Screen with what appears to be a Micro SD Card Symbol at the bottom of the screen. The screen also appears to "brighten" it's blacklight on and off every 2 seconds, and .fails to go any further.

I have rebooted the printer several times now with no result.

I expected the screen to display it's usual start up message and run the XTouch program.

[image]

**XTOUCH (please complete the following information):**
 - FW: Unsure, but was the latest downloaded on 17/4/2025

Printers 
 - Device: P1S
 - Printer FW Version: 01.07.00.00
-  AMS units: 1
-  AMS FW: 00.00.06.49

JesterITA, 2025-04-17:
Same behavior here.
OTA Update at boot and stuck on this screen.

P1S 
FW 01.07.00.00
No AMS

xperiments, 2025-04-17:
We have moved to LanOnly Mode

https://github.com/xperiments-in/xtouch?tab=readme-ov-file#installation

closed 2025-04-17 by xperiments