xperiments-in/xtouch #203: Flashing screen with SD card sign after update

stevvie, 2025-04-19:
Just turned the Xtouch on and it did an update and then verified and installed the firmware and now just stuck at a flashing screen every 3 seconds and small sd card icon in the bottom centre of the screen. Taken the SD card out and put into another Xtouch that has updates turned off and it works just fine. Tried the card from the other xtouch with old firmware (0.9.147) in the newly flashed device and same flashing screen

**To Reproduce**
turn on device after the update.

**Expected behavior**
working xtouch

**Screenshots**

[image]
**XTOUCH (please complete the following information):**
 - FW: [Unknown as no longer booting]

**Printers (please complete the following information):**
 - Device: [P1S]
 - FW Version [1.07.00.00]
-  AMS units:[1]

stevvie, 2025-04-19:
Managed to fix it. 
Just needed a new xtouch.json generated from https://xperiments.in/xtouch-bin/localOnly.html
I must of totally missed the message that said it needs a new version of xtouch.json when it had updated

BlackCatPeanut, 2025-04-19:
Thanks for coming back and updating the issue. I did the same thing as above and it was resolved.

ploonnl, 2025-04-19:
Thanks, that does work!

gusosborne, 2025-05-14:
Same issue here. I would call this a "breaking update". It would be really nice to have a warning message that informs the user that their working xtouch device will become non-functional, until they figure this out.

lucacri, 2025-05-31:
That is such an irresponsible update! It's a massive breaking change, and nothing was done to prevent/inform (it could have been a major version, a note when updating "hey it's gonna screw up all of it", anything).

Not even a migration guide (which at least would show that there is the need for a migration in the first place).

Seriously, this is just amateur hour

closed 2025-06-09 by xperiments