xperiments-in/xtouch #206: Temperature set doesn't work

spedruzzo, 2025-04-30:
**Describe the bug**
X-touch connected, configured, axis movement works, fan control works bur unload load filament doesn't work and temperature set doesn't work

**To Reproduce**
Steps to reproduce the behavior:
1. Go to temperature control 
2. Click on temperature set to 250 or any temp
3. hit enter
4. the extruder icon change to orange for a second then return green without changing the temperature

**Expected behavior**
the icon became orange and temperature start rising

**XTOUCH (please complete the following information):**
 - FW: 0.9.212

**Printers (please complete the following information):**
 - Device: P1S
 - FW Version: 01.07.00.00

xperiments, 2025-04-30:
Read new docs. We moved to lanonly mode

xperiments, 2025-04-30:
> AH ok yes I try and it works (not so clear in doc but maybe I miss the
> section)
> anyway this is completely useless because I loose the android app feature
> why this behaviour? is not possible to enable a different authentication?
> thanks for quick answer

Ask BBL not me xd
They forced US to move to lanonly devmode

Btw i am preparing a thing covers 90% usage of y android app

If you use It to print directly models from makerworld them thas the 10%
missing

Live view IS implemented

spedruzzo, 2025-04-30:
:) I understand 
probably BBL prefer the 70 euro BIGTREETECH Panda Touch (I suppose it works due to an accord with BBL)

spedruzzo, 2025-04-30:
Anyway thanks for this great work

xperiments, 2025-04-30:
No, panda touch IS same problem

closed 2025-06-04 by xperiments