xperiments-in/xtouch #210: Can't control my P1S with the screen, can only show the info

lucky13820, 2025-06-09:
**Describe the bug**
I can't unload or load filament, change temperature. No control works, but all the info are updating. I tried to reinstall the firmware by connecting to my laptop and install with the page. https://xperiments.in/xtouch-bin/

**XTOUCH (please complete the following information):**
 - FW: 0.9.212

**Printers (please complete the following information):**
Device: [P1S]
FW Version [1.07.00.00]
No AMS

**Additional context**
Add any other context about the problem here.

xperiments, 2025-07-21:
Aside your printer version xtouch only supports lanmode now

MStampka, 2025-10-03:
Describe the bug
i have the same problem as in the fist post.
I've reinstalled the x-touch FW and set the printer to LAN only, without any changes.

XTOUCH (please complete the following information):
FW: 0.9.212

Printers (please complete the following information):
Device: [P1S]
FW Version [1.09.00.00]